aschulm/battor #39: `make install` hangs on `battor -k`

gurupras, 2016-06-02:
I've been running an old version of the BattOr firmware -  

I recently tried to update the BattOr firmware to a newer revision - 667213f. This failed with an error stating that `/dev/ttyUSB0` could not be found. Thus, I tried to dig into this to identify the commit that changed the Makefile in some way. During `git bisect`, while I was compiling and flashing the firmware present in different commits, one of the commits caused the BattOr to stop responding. I'm not sure which commit this was.

Currently, This is what I get:
    avrdude -e -p x192a3u -c avr109 -b 115200 -P /dev/ttyUSB0 -U flash:w:main.hex
    avrdude: butterfly_recv(): programmer is not responding

Enabling verbose output, I get:
    avrdude -e -p x192a3u -c avr109 -b 115200 -P /dev/ttyUSB0 -U flash:w:main.hex -v -v -v -v -v
    avrdude: Version 6.0.1, compiled on Oct 21 2013 at 15:55:32
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2009 Joerg Wunsch
    
             System wide configuration file is "/etc/avrdude.conf"
             User configuration file is "/home/guru/.avrduderc"
             User configuration file does not exist or is not a regular file, skipping
    
             Using Port                    : /dev/ttyUSB0
             Using Programmer              : avr109
             Overriding Baud Rate          : 115200
             AVR Part                      : ATxmega192A3U
             Chip Erase delay              : 0 us
             PAGEL                         : P00
             BS2                           : P00
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 0
             StabDelay                     : 0
             CmdexeDelay                   : 0
             SyncLoops                     : 0
             ByteDelay                     : 0
             PollIndex                     : 0
             PollValue                     : 0x00
             Memory Detail                 :
    
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
                 Memory Ops:
                   Oeration     Inst Bit  Bit Type  Bitno  Value
                   -----------  --------  --------  -----  -----
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               prodsig        0     0     0    0 no         50   50      0     0     0 0x00 0x00
                 Memory Ops:
                   Oeration     Inst Bit  Bit Type  Bitno  Value
                   -----------  --------  --------  -----  -----
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
                 Memory Ops:
                   Oeration     Inst Bit  Bit Type  Bitno  Value
                   -----------  --------  --------  -----  -----
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
                 Memory Ops:
                   Oeration     Inst Bit  Bit Type  Bitno  Value
                   -----------  --------  --------  -----  -----
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
[... 56 lines cut ...]
                 Memory Ops:
                   Oeration     Inst Bit  Bit Type  Bitno  Value
                   -----------  --------  --------  -----  -----
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               usersig        0     0     0    0 no        512  512      0     0     0 0x00 0x00
                 Memory Ops:
                   Oeration     Inst Bit  Bit Type  Bitno  Value
                   -----------  --------  --------  -----  -----
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
                 Memory Ops:
                   Oeration     Inst Bit  Bit Type  Bitno  Value
                   -----------  --------  --------  -----  -----
    
             Programmer Type : butterfly
             Description     : Atmel AppNote AVR109 Boot Loader
    
    Connecting to programmer: .avrdude: Send: . [1b] 
    avrdude: Send: S [53] 
    avrdude: ser_recv(): programmer is not responding
    avrdude: butterfly_recv(): programmer is not responding

csarron, 2016-12-08:
Same problem here

aschulm, 2017-06-19:
Sorry for the extreme delay in replying to this bug. I must have missed the email notification for it.

I'm doing a sweep to close any old bugs that remain open. Are either of you still having this issue?

It sounds to me like you might have gotten your BattOr stuck on a version of the firmware that did not have the ability to reset via a serial command. This reset is needed to enter the Bootloader (indicated by the quick flashing yellow light).

You can try to flash the BattOr to the latest version of the firmware by commenting out `battor -k`, the resetting the BattOr manually (open the BattOr's plastic enclosure and hit the reset button), and quickly after you hit the button running `make flash`.