aschulm/battor #53: GET_SAMPLE_COUNT can result in a halt

aschulm, 2016-08-17:
There are two issues here, the first is when the device is buffering, any incoming bytes will interrupt the processor and take up precious CPU cycles that are needed for handling the incoming stream of samples. This could lead to various halt conditions. 

The problem is we want the client to be able to send a clock sync whenever it likes, not knowing what the BattOr is doing, and the client wants the BattOr to respond as quickly as possible. This means there will always be a chance that a clock sync message DOSes the BattOr.

aschulm, 2016-08-19:
The attached trace shows the BattOr resetting when a SAMPLE_COUNT control message is received. Again likely due to the BattOr still sampling and the DMA buffers overflowing when TX'ing the sample count over UART goes out to lunch.

CC @testaco 

[battor-serial-log-2016-08-18_19-32.txt-24327.txt](https://github.com/aschulm/battor/files/426269/battor-serial-log-2016-08-18_19-32.txt-24327.txt)

aschulm, 2016-08-29:
ok so this issue is more subtle than I thought. I actually was misinterpreting the logs, the battor is actually resetting before the record clock sync is sent, and the reset byte is seen only when the read is called for the record clock sync. I figured this out by standing next to the battor when I ran a trace. it also didn't help to unplug and replug the power.

I added debug prints to the firmware to figure out why it is resetting, (will attach soon) and it turns out to be the sram is filling up, that's why we are seeing error 12, RINGBUF WRITE FAIL. that only happens when SD card writes are going very slowly. so I added more debug prints and confirmed that most sd card writes are taking hundreds of milliseconds. 

the reason seems to be because we are using single block writes and sd cards are optimized for multi block writes. so we need to modify the filesystem to have it do multi block writes with the multi block pre-erase command before hand.

testaco, 2016-08-31:
Follow-up to Aaaron's assessment: I'm currently working on a system-level test that will confirm and reproduce the issue as described.

Adding multi block write and pre-erase is probably a smart feature to add no matter what, but I want to make sure that we fully characterize and solve this immediate issue (the symptom being that the battor resets after 6 second of recording samples).

testaco, 2016-09-01:
I'm working on a branch that will add a new debugging feature: if you add DEBUG_SD_TIMING to the firmware build, it will log the average block write time periodically.  I will then collect a graph of average write times as I stress test a BattOr and confirm that eventually it reaches a point where write lag outpaces the sample rate.

Then we can add multi-block write and show that average write times does not increase.  If there are any other issues with SD card writing efficiency this feature should help tease them out.

aschulm, 2016-09-01:
you actually should be able to print every sd card write if that's easier. there is a global in the timer that keeps track of milliseconds for timestamping

testaco, 2016-09-07:
Here's a plot of the bad SD card's block write times (horizontal axis is block number, vertical axis is in milliseconds): [image]

Here's a plot of the bad SD card after it has worked through the slow section: 
[image]

Notice that when the SD card is in the bad mode, it frequently needs to pause for 120ms+.  This repeated need for long block write times eventually causes the SRAM to fill up.  This causes the device to reset with error 12 (out of space in SRAM).

Here's the PR that instruments write timing analysis for SD cards as a debugging option: https://github.com/aschulm/battor/pull/54

It looks like the best bet is still to implement multi-block write so that way we give the SD card the best chance possible to catch up if it falls behind the ringbuffer.

testaco, 2016-09-20:
I've got the branch testa/fs-stream now working using multi-block write.  Take a look at this capture from the SD card referenced in my last post, but using multi-block write.  It's much much faster!

[image]

I'll be testing the firmware with this change and then merge into master once we're sure there's no regressions.

testaco, 2016-10-14:
Should be fixed by #56 which is now on master...

closed 2017-01-30 by aschulm